microsoft/AirSim #941: Hexacopter Setup in Docs and MultiRotorParams seems a little off

benasena, 2018-04-07:
I noticed that the generation of motors in MultiRotorParams and the corresponding documentation might be a bit off. If I'm reading this correctly, the quaternions rotate the arm vectors always CCW by 30 degrees (pi/6), so motors 4 and 5 (zero-indexed) are 30 degrees off the East axis instead of 60 like I remember hexagons looking like.

I defined two different quaternions to solve this and keep the syntax the same:
Quaternionr hexax_rot_30(AngleAxisr(M_PIf / 6, unit_z));
Quaternionr hexax_rot_60(AngleAxisr(M_PIf / 3, unit_z));

Do let me know if I'm completely off-base somehow.

https://github.com/Microsoft/AirSim/wiki/hexacopter
https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp

benasena, 2018-04-09:
I also noticed that FlyingPawn.h hard-codes rotor count as 4. Not sure if this modification or another workaround was forgotten in the write-up or if something about the implementation changed in the meantime, but it was necessary for me to update rotor_count = 6 to get the extra two props to animate.

https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Multirotor/FlyingPawn.h

(I also noticed you can change the direction of the prop Rotation objects in the Unreal blueprints by setting Plane Constraint Normal Z to -1.)

sytelus, 2018-04-09:
@clovett might be interested in this.

lovettchris, 2019-04-16:
I have some fixes and a demo video in this pull request: https://github.com/Microsoft/AirSim/pull/1890

manthan99, 2020-05-22:
@lovettchris is it possible to run the Hexacopter in a SITL mode or it is only supported for HIL

jonyMarino, 2020-10-19:
Issue solved. Doc update pending

closed 2020-10-19 by jonyMarino