microsoft/AirSim #944: simGetImages memory error

M-Kasem, 2018-04-08:
I have a script that collects images in cvmode. The memory allocated to the python process grows overtime due to calling simGetImages multiple times and eventually the process terminates. I have added a stress test script that just calls simGetImages. I am monitoring the process's memory using System Monitor in Ubuntu 16.04.

```
import time
from AirSimClient import *
client = MultirotorClient(ip = "", port= 41451)
client.confirmConnection()
for i in range(200000):
    responses = client.simGetImages([ImageRequest(0, AirSimImageType.Scene, False, False)])
    time.sleep(0.01)
```
The problem becomes much worse with higher resolution images, I am using "Width": 1024 and "Height": 576.

Is there a workaround to avoid this memory problem?

sytelus, 2018-04-09:
Thanks for reporting this. We will be looking in to this but if you have time, any hints would be helpful as well :).

henrik94, 2018-07-06:
Is there any update on this? We also experienced growing memory usage after calling the getImages function in PawnSimApi.cpp several times / sec.

jonyMarino, 2020-08-14:
I close this due to age. If still happens on the new version Please feel free to open a new issue and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-08-14 by jonyMarino