microsoft/AirSim #946: Values being sent for API

NextSim, 2018-04-09:
I am working on using the API and am able to send data to control the car.  There are two values I don't understand though.  To enable API control, the helloCar example sends the data below (in JSON)

> 
> [
> 0,
> 1,
> "enableApiControl",
> [
> true
> ]
> ]

Where do the 0 and 1 come from?

When it sends the car control to set the throttle, etc, it sends 0 and 3.  Where do these values come from?

sytelus, 2018-04-09:
Are you using direct json? I think you want to use PythonClient or C++ wrapper classes that internally serializes to json so you don't have to worry about. Above might be fields used by msgpack-rpc protocol but I haven't looked in to it.

sytelus, 2018-04-12:
wow... that's a very long way to make this work ;). You should just use documented way to use this library. See https://github.com/yfakariya/msgpack-rpc-cli/wiki/Quick-start

sytelus, 2018-04-12:
Let's continue discussion over to https://github.com/Microsoft/AirSim/issues/876

closed 2018-04-12 by sytelus