microsoft/AirSim #947: LINK : fatal error LNK1257: 代码生成失败 [F:\AirSim\HelloCar\HelloCar.vcxproj]

JLUBMH, 2018-04-09:
f:\airsim\hellocar\main.cpp : fatal error C1001: 编译器中发生内部错误。 [F:\AirSim\HelloCar\HelloCar.vcxproj]
  (编译器文件“f:\dd\vctools\compiler\utc\src\p2\p2symtab.c”，第 2285 行)
   要解决此问题，请尝试简化或更改上面所列位置附近的程序。
  请选择 Visual C++
  “帮助”菜单上的“技术支持”命令，或打开技术支持帮助文件来获得详细信息。
LINK : fatal error LNK1257: 代码生成失败 [F:\AirSim\HelloCar\HelloCar.vcxproj]
已完成生成项目“F:\AirSim\HelloCar\HelloCar.vcxproj”(默认目标)的操作 - 失败。

已完成生成项目“F:\AirSim\AirSim.sln”(默认目标)的操作 - 失败。

sytelus, 2018-04-09:
Hello, unfortunately we can't read Chinese. Your help in translating and debugging would be appreciated :).

clovett, 2018-04-09:
F:airsimhellocarmain.cpp:fatal error C1001: Internal errors occurred in the compiler.
[F:AirSimHelloCarHelloCar.vcxproj]
(Compiler file "F:DDVCTOOLSCOMPILERUTCSRCP2P2SYMTAB.C", line No. 2285)
To resolve this issue, try simplifying or changing the program near the location listed above.
Please select Visual C + +
The Technical Support command on the Help menu, or open the technical Support help file for more information.
Link:fatal Error LNK1257: code generation failed [F:AirSimHelloCarHelloCar.vcxproj]
Completed the operation of Build project ' F:AirSimHelloCarHelloCar.vcxproj ' (default target)-failed. Completed the operation of Build project ' F:AirSimAirSim.sln ' (default target)-failed.

JLUBMH, 2018-04-10:
F:airsimhellocarmain.cpp:fatal error C1001: Internal errors occurred in the compiler.
[F:AirSimHelloCarHelloCar.vcxproj]
(Compiler file "F:DDVCTOOLSCOMPILERUTCSRCP2P2SYMTAB.C", line No. 2285)
To resolve this issue, try simplifying or changing the program near the location listed above.
Please select Visual C + +
The Technical Support command on the Help menu, or open the technical Support help file for more information.
Link:fatal Error LNK1257: code generation failed [F:AirSimHelloCarHelloCar.vcxproj]
Completed the operation of Build project ' F:AirSimHelloCarHelloCar.vcxproj ' (default target)-failed. Completed the operation of Build project ' F:AirSimAirSim.sln ' (default target)-failed.

sytelus, 2018-04-12:
Sorry, this looks like error specific to your VS install. Please make sure you have VS2015 Update 3  and you are running from VS2005 x64 Command Prompt as described in build doc. As we don't have Chinese version of OS, you might have to help us debug this and contribute any potential change.

closed 2018-04-12 by sytelus

bys1123, 2018-04-12:
Hey, I'll invite you join Chinese AirSim and Pixhawk user group. Add my wechat微信 253331754.