microsoft/AirSim #949: Unstable PX4 in AirSim

ypv, 2018-04-10:
Hi,

I have been testing PX4 SITL in AirSim for some time now in windows/linux and I can't get it to be stable either with v1.5.5 as recommended in readme or any of the later version all the way to the latest. It appears that simple flight is the answer by the team right now to ensure drone stability. However we have migrated from gazebo to airsim and would like PX4 to work. There was a mavlink-centric discussion mentioning instability caused by gravity/zacc and HIL_STATE_QUATERNION vs HIL_SENSOR and proposed fix. Is this fix worth pursuing on our own? Is the team not interested in PX4 autopilot stability? 

Please comment and propose a way forward,
ypv

lovettchris, 2018-04-11:
I've been flying PX4 (Pixhawk, Pixracer, Pixfalcon and Pixhawk 2.1) all in HIL mode using latest PX4 bits (1.7.3) and it flies great.  So this must be a SITL only problem.

benasena, 2018-04-12:
I've been using PX4 SITL with some mixed results as well, but have been able to get some stable flying out of it. Some useful pxh commands for debugging are:

commander status
gpssim status
ekf2 status

And if the estimator gets screwy you can try 

ekf2 stop
ekf2 start

And wait for a new ekf2 alignment.

ypv, 2018-04-12:
Hi Chris,
You are right. I have not tested HITL PX4+airsim. I have only flown SITL both in windows and linux and there is consistent failure to be stable. It does get worse if you do not reset PX4 state. Windows tends to do slightly better than Linux. 

It is interesting that HITL is working great for you. I am not sure what that means for getting SITL to work, but I will try to find out. 

One interesting conclusion that I am coming to is that if HITL is interfacing with airsim well, then this SITL issue can be found in the SITL-only part of the PX4 stack.

lovettchris, 2018-12-18:
A bunch of bug fixes have been made in AirSim and PX4 SITL and HITL both work great with PX4 version v1.8.2.

closed 2018-12-18 by madratman

machenxiang, 2019-03-18:
@lovettchris hi cris. I’m sorry to bother you. But my current master and any version of px4 firmware can’t work fine . The drone can’t fly stable. Can you tell me how to solve this issue?

lovettchris, 2019-03-20:
please open a new issue and I'll take a look at it there.