microsoft/AirSim #950: Error in ./build.sh

ChampDBG, 2018-04-10:
Hi, I was executing the script './build.sh' and got the following error.

> [ 85%] Building CXX object AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/home/wu/AirSim/AirLibUnitTests/main.cpp.o
In file included from /home/wu/AirSim/AirLibUnitTests/main.cpp:3:
In file included from /home/wu/AirSim/AirLibUnitTests/RosFlightTest.hpp:5:
In file included from /home/wu/AirSim/AirLib/include/vehicles/multirotor/MultiRotorParamsFactory.hpp:7:
In file included from /home/wu/AirSim/AirLib/include/vehicles/multirotor/configs/Px4MultiRotor.hpp:8:
/home/wu/AirSim/AirLib/include/common/AirSimSettings.hpp:464:28: error: no member named
      'insert_or_assign' in 'std::__1::map<std::__1::basic_string<char>,
      msr::airlib::AirSimSettings::PawnPath, std::__1::less<std::__1::basic_string<char> >,
      std::__1::allocator<std::__1::pair<const std::__1::basic_string<char>,
      msr::airlib::AirSimSettings::PawnPath> > >'
                pawn_paths.insert_or_assign(key,
                ~~~~~~~~~~ ^
1 error generated.
AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/build.make:62: recipe for target 'AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/home/wu/AirSim/AirLibUnitTests/main.cpp.o' failed
make[2]: *** [AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/home/wu/AirSim/AirLibUnitTests/main.cpp.o] Error 1
CMakeFiles/Makefile2:328: recipe for target 'AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/all' failed
make[1]: *** [AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

ChampDBG, 2018-04-10:
I looked the modified history about ['AirSimSettings.hpp'](https://github.com/Microsoft/AirSim/commit/b8ed712cc2da1707787236d98985d985a12a5053#diff-4ee334619d0e207657a29f2139412385) and executed successfully with changing 'insert_or_assign' to 'emplace' again.

The command 'insert_or_assign' and 'emplace' seem belong to different C++ version (insert_or_assign is C++ 17 and emplace is C++11). Unfortunately, I am not familiar with C++, so I have no idea to deal with this problem with other way.

vkotaru, 2018-04-11:
same error

vkotaru, 2018-04-11:
Tried changing the compiler to `g++-7` in (line  35) build.sh, but it did not work.

sytelus, 2018-04-12:
This is fixed now. Thanks for reporting.

closed 2018-04-12 by sytelus