microsoft/AirSim #951: #951 downloading rpclib：拒绝访问

XueYuSky, 2018-04-10:
win10,VS2015 pro3,  cmake 3.9.3
when I run build.cmd,err occurs:

C:\Users\XueYuSky\source\repos\AirSim>build.cmd
Found cmake version: 3.9.6

*****************************************************************************************
Downloading rpclib
*****************************************************************************************
拒绝访问。
拒绝访问。
找不到 C:\Users\XueYuSky\source\repos\AirSim\external\rpclib.zip
Unable to download high-polycount SUV. Your AirSim build will use the default vehicle.

#### Build failed - see messages above.

XueYuSky, 2018-04-10:
English:

C:\Users\XueYuSky\source\repos\AirSim>build.cmd
Found cmake version: 3.9.6

Downloading rpclib

 access denied.
 access denied.
Can't find  C:\Users\XueYuSky\source\repos\AirSim\external\rpclib.zip
Unable to download high-polycount SUV. Your AirSim build will use the default vehicle.

Build failed - see messages above.

XueYuSky, 2018-04-10:
reasong: I didn't restart my PC after I changed the enviroment variable.

I reboot my  computer and build again, it works. then encounter the half.h issues which someone have solve that problem.

sytelus, 2018-04-12:
I can't reproduce this issue. I'm guessing you have some permission issues. We'll close this issue for now.

closed 2018-04-12 by sytelus

bys1123, 2018-04-12:
Chinese Internet blocked rpclib download sometime.

XueYuSky, 2018-04-15:
Thank, I have solved this problem.

hassanyf, 2018-06-20:
How did you solve it?

futuretx, 2019-01-11:
I met the same err, would you show the reason?

XueYuSky, 2019-01-23:
> How did you solve it?

> How did you solve it?

> I met the same err, would you show the reason?
I used my sserver in Singapore,recompile it.

calmelo, 2020-07-16:
把杀毒软件关了就可以了