microsoft/AirSim #954: How can I modify the name of images collected from recording?

BrianN92, 2018-04-11:
After recording, I got image names like "img_0_0_341337095290.png". I am wondering how can I modify their name, for example change to "img_0.png", "img_1.png", etc.  Then, the "ImageName" tag inside "airsim_rec" could also be modified. Thank you!

sytelus, 2018-04-12:
Currently you will need to modify this code: https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Recording/RecordingFile.cpp#L22

Please feel free to contribute any enhancements to do this from config.

closed 2018-04-12 by sytelus

BrianN92, 2018-04-12:
Hi @sytelus, thanks for your advice! I tried to modify the code you marked, changed it to"image_file_name << "img_" << i <<(response.pixels_as_float ? ".pfm" : ".png");", (I want the output like "img_0", "img_1", etc.) and then record. The names of images still stayed unchanged. I have to admit that I have little knowledge about c++ language. How can I deal with it? Thank you!

I am using Windows 10 as operating system.

kutilanak, 2022-04-10:
> Hi @sytelus, thanks for your advice! I tried to modify the code you marked, changed it to"image_file_name << "img_" << i <<(response.pixels_as_float ? ".pfm" : ".png");", (I want the output like [...]

Hello, have you find the solution? thank you