microsoft/AirSim #955: error when calling client.getMultirotorState() in hello_drone.py

zhonghankb, 2018-04-11:
If you are looking for a new feature or bug fix, fire up your IDE, try playing around with source code and contribute back! 
What's better than filing issue? Filing a pull request :).

---- (Remove above before filing the issue) -----
Hi all 
I am new to AirSim and python. When I was trying to run the hello_drone.py with the Neighbourhood  Unreal environment, there was error as below:
######################
Waiting for connection:
Traceback (most recent call last):
  File "hello_drone.py", line 18, in <module>
    state = client.getMultirotorState()
  File "C:\Users\kobe\Documents\github\projects\airsim\AirSim\PythonClient\AirSimClient.py", line 523, in getMultirotorState
    return MultirotorState.from_msgpack(self.client.call('getMultirotorState'))
  File "C:\ProgramData\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\ProgramData\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'getMultirotorState' with argument count 0.
##########################
Configuration:
AirSim: built with VS2015 update 3 on Mar.31 2018
OS: windows 10 x64
pyhton: 3.6.4

Hope someone could help me with it.

sytelus, 2018-04-12:
Just tested this with most recent released binary and it seems to work fine. I'm guessing you are using old binary or old PythonClient.

zhonghankb, 2018-04-14:
I have updated both the PythonClient and the binary, now it works fine, thanks.

closed 2018-04-14 by zhonghankb