microsoft/AirSim #956: How to set time of day for the evening?

dehuachen2017, 2018-04-12:
I want drones to fly in the evening. In file settings.json, I set

"TimeOfDay": {
"Enabled": true,
"StartDateTime": "%2018-%04-%12 %18:%30:%00"
}

But the drone still flies during the day. How to set time of day for the evening? Thanks.

sytelus, 2018-04-12:
I think you want:

```
"TimeOfDay": {
  "Enabled": true,
  "StartDateTime": "2018-04-12 18:30:00"
}
```

dehuachen2017, 2018-04-12:
Thank you for your response.  I use  "StartDateTime": "2018-04-12 20:00:00" in file settings.json. The scene does not change. The drone still flies during the daytime. Can you help me? Thanks.

sytelus, 2018-04-13:
It's working for me:

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "TimeOfDay": {
     "Enabled": true,
     "StartDateTime": "2018-04-12 18:30:00"
  }
}
```

[image]

fshamshirdar, 2018-11-09:
Any update on this? it uses the current time not the StartDateTime.

bmalhigh, 2018-11-27:
This a bug. Working on fix.

bmalhigh, 2018-11-28:
This should be fixed now with the pull request 1581

closed 2018-11-28 by bmalhigh