microsoft/AirSim #959: Is SimSteering (Steering wheel + Pedals) compatible with AirSim?

haider04, 2018-04-13:
Hi,
I am working on a vehicle simulator for research purposes and it has SimSteering components (Steering wheel + brake, accelerator pedals) integrated within the vehicle simulator. I wanted to know if SimSteering configuration (Advance force feedback system) is compatible with AirSim and car can be controlled without any problems in windows environment? 
Any help in this regard will be highly appreciated. Thanks!

Haider

sytelus, 2018-04-14:
We unfortunately don't have experience with SimSteering system but it should be possible to interface it with AirSIm. You can look at SimJoyStick folder to see how joystick is interfaced with AirSim.

jonyMarino, 2020-08-13:
Closing because this is out of the project scope

closed 2020-08-13 by jonyMarino