microsoft/AirSim #961: Camera ID doesn't work for right and left on multirotor

Pierrolo, 2018-04-13:
Hi, 
I have the followings setting:

    "SubWindows": [
    {"WindowID": 0, "CameraID": 1, "ImageType": 3, "Visible": true},
    {"WindowID": 1, "CameraID": 0, "ImageType": 3, "Visible": true},
    {"WindowID": 2, "CameraID": 2, "ImageType": 3, "Visible": true}
    ]

And this is what I get:

 
[image]
As you can see, no matter the camera_id, it always give me the front one...

Same goes when I am requesting images throught the python API : Camera_id 0,1 and 2 give the same images, when I actually want one front, one right and one left. 

Also, it seems the depth is calculated on object, not on pixel, in the picture above, if you only look at the depth images, you would think there is a wall in front of me, when really it's just the floor. But the floor 1cm in front of me and the floor 2m away from me, should not give the same depth value.... Right?

The `FOV_Degrees` setting doens't work etiher, whatever the value I give it, it also shows only for 90degrees..

Thanks for your help

sytelus, 2018-04-14:
Camera 0,1,2 are very close so you might not see lot of difference in images. Their use is mainly for stereo algorithms. Have you tried camera 3,4,5?

FOV setting also works.

This is what I get for below settings:

```
  "SubWindows": [
	{"WindowID": 0, "CameraID": 2, "ImageType": 0, "Visible": true},
	{"WindowID": 1, "CameraID": 3, "ImageType": 1, "Visible": true},
	{"WindowID": 2, "CameraID": 4, "ImageType": 3, "Visible": true}
  ],
  "CaptureSettings": [
    {
      "ImageType": 0,
      "FOV_Degrees": 130
    }
  ]
```

[image]

suhalvemu, 2018-04-14:
E
Thanks,
Suhal Vemu.

closed 2020-02-14 by ironclownfish