microsoft/AirSim #966: How to display the picture acquired by the camera in real time （with opencv）

koalamaster, 2018-04-14:
## **By converting vector<uint8_t> to Mat, I cannot use opencv to display the image correctly. I don't know where I am wrong.**

```
int getOneImage()
{
	using std::cin;
	using std::cout;
	using std::endl;
	using std::vector;
	using cv::imshow;
	using cv::Mat;
	using cv::waitKey;
	using namespace msr::airlib;

	typedef ImageCaptureBase::ImageRequest ImageRequest;  // camera_id, image_type, pixels_as_float_val(false), compress_val(true)
	typedef ImageCaptureBase::ImageResponse ImageResponse;
	typedef ImageCaptureBase::ImageType ImageType;

	msr::airlib::MultirotorRpcLibClient client;
	client.confirmConnection();  // check connection	
	
	vector<ImageRequest> request = { ImageRequest(0, ImageType::Scene) };
	const vector<ImageResponse>& response = client.simGetImages(request);

	cout << "the size of images recieved: " << response.size() << endl;
	if (response.size() > 0)
	{    
                // create Mat to save img
		cv::Mat camera_0_img = cv::Mat(response.at(0).height, response.at(0).width, CV_8UC3, cv::Scalar(0, 0, 0));

		//vector<uint8_t> camera_0_img_data = response.at(0).image_data_uint8;
		//cout << "camera_0_img_data.size(): " << camera_0_img_data.size() << endl;

		int camera_0_img_width = response.at(0).width;
		int camera_0_img_height = response.at(0).height;

                //  vector<uint8_t> to Mat
		for (int row = 0; row < response.at(0).height; row++)
		{
			for (int col = 0; col < response.at(0).width; col++)
			{
				camera_0_img.at<cv::Vec3b>(row, col) = cv::Vec3b( 
response.at(0).image_data_uint8[row*camera_0_img_width + 3*col + 0], 
response.at(0).image_data_uint8[row*camera_0_img_width + 3*col + 1],  
response.at(0).image_data_uint8[row*camera_0_img_width + 3*col + 2]);													
			}
		}
		imshow("camera_0_img", camera_0_img);
		waitKey(0);
	}
	return 0;
}
```

aburgm, 2018-04-19:
If you are reading the pixel data like this, make sure to request the image in uncompressed format, i.e. set the third parameter of `ImageRequest` to false. I think it is true (compressed) by default. Also, you are getting an RGBA image (4 channels).

pkicman, 2018-04-20:
Please, also remeber that by default OpenCV uses BGR order of colour channels, so you need to correctly assign them to your output image.

sytelus, 2018-04-25:
See here for the numpy conversion code: https://github.com/Microsoft/AirSim/blob/master/PythonClient/hello_car.py#L64

wujiyoung, 2018-05-08:
@allen9273, I load the image data to opencv Mat by `cv::Mat im = cv::imdecode(image_info.image_data_uint8, CV_LOAD_IMAGE_GRAYSCALE);`, it works perfectly for me.

koalamaster, 2018-05-10:
> Have you successfully  display the images?  Could you share the code to me? Because I am struggling with this as well.

font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    

Mat image = cv::imdecode(image_info.image_data_uint8, cv::IMREAD_COLOR);

    
        a#ntes-pcmail-signature-default:hover {
            text-decoration: underline;
            color: #199cff;
            cursor: pointer;
        }
        a#ntes-pcmail-signature-default:active {
            text-decoration: underline;
            color: #246fce;
            cursor: pointer;
        }
    

                
                    发自网易邮箱大师

damien022, 2018-12-12:
this code works for me although there must be faster ways to copy the data across, note that I am asking for uncompressed data, and I am copying the data as BGRA (OpenCV expected format)

```
int getoneImage()
{
    using std::cin;
    using std::cout;
    using std::endl;
    using std::vector;
    using cv::imshow;
    using cv::Mat;
    using cv::waitKey;
    using namespace msr::airlib;

    typedef ImageCaptureBase::ImageRequest ImageRequest;  // camera_id, image_type, pixels_as_float_val(false), compress_val(true)
    typedef ImageCaptureBase::ImageResponse ImageResponse;
    typedef ImageCaptureBase::ImageType ImageType;

    msr::airlib::MultirotorRpcLibClient client();
    client.confirmConnection();  // check connection
    // note the second false is to ask for uncompressed data
    vector<ImageRequest> request = { ImageRequest("0", ImageType::Scene, false, false) };  
  bool exit = false;

    while (!exit) {
        const vector<ImageResponse>& response = client.simGetImages(request);
        if (response.size() > 0)
        {
            cout << "raw image response: " << response.at(0).image_data_uint8.size() << endl;
            // create Mat to save img
            cv::Mat camera_0_img = cv::Mat(response.at(0).height, response.at(0).width, CV_8UC4, cv::Scalar(0, 0, 0, 0));

            //vector<uint8_t> camera_0_img_data = response.at(0).image_data_uint8;
            //cout << "camera_0_img_data.size(): " << camera_0_img_data.size() << endl;

            int camera_0_img_width = response.at(0).width;
            int camera_0_img_height = response.at(0).height;

            //  vector<uint8_t> to Mat
            for (int row = 0; row < response.at(0).height; row++)
            {
                for (int col = 0; col < response.at(0).width; col++)
                {
                    camera_0_img.at<cv::Vec4b>(row, col) = cv::Vec4b(
                            response.at(0).image_data_uint8[row*camera_0_img_width*4 + 4*col + 2],
                            response.at(0).image_data_uint8[row*camera_0_img_width*4 + 4*col + 1],
                            response.at(0).image_data_uint8[row*camera_0_img_width*4 + 4*col + 0],
                            response.at(0).image_data_uint8[row*camera_0_img_width*4 + 4*col + 3]);
                }
            }
            imshow("camera_0_img", camera_0_img);
            waitKey(10);

        }
    }
    return 0;
}
```

closed 2019-10-03 by msb336