microsoft/AirSim #974: Steering wheel with Airsim

BrianN92, 2018-04-16:
"PhysXCar" : {
    "RC": {
        "RemoteControlID": 0
    }
}

After adding above settings to the json file, a wired thing happened. I can only control the direction of car by accelerator pedal. The steering angle was -1.25 when I release the pedal, and the value would change with the change of force I gave to the pedal. Did anyone meet this kind of issue? Thank you!

BrianN92, 2018-04-16:
BTW, I'm using Logitech G29 as a steering wheel controller.

BrianN92, 2018-04-16:
Besides, I did put the x360ce exe in the same directory as the AirSim binary.

mitchellspryn, 2018-04-24:
For the record, we solved the problem in [this issue](https://github.com/Microsoft/AutonomousDrivingCookbook/issues/27). tl;dr: if you're using the x360ce route, make sure the x360ce executable is run from the **same directory** as the binary. This might be nested within subfolders.

sytelus, 2018-04-25:
FYI - AirSim now has out-of-support for Logitech G920 wheel (i.e. don't put x360ce in folder). We don't have G29 so we haven't tested that one but you can try remove x360ce  and see if that works. If it doesn't, it might be just matter of [extending this code](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp#L68) a little bit.

rohjunha, 2018-04-26:
I am currently using G29 and code below worked for me with a PS3 setup:
```
joystick_controls_.steering = joystick_state_.left_x;
joystick_controls_.throttle = joystick_state_.left_y;
joystick_controls_.brake = (joystick_state_.left_y < -0.5f) ? 1.0f : 0.0f;
```

sytelus, 2018-04-27:
Would you mind to get pull request for this?

BrianN92, 2018-04-30:
Hi @mitchellspryn , did you try to use Thrustmaster in the new released city map? I [failed to do that.](https://github.com/Microsoft/AirSim/issues/1024)

mitchellspryn, 2018-05-01:
Nope. It was broken for me, too (also for Hawaii). I haven't gotten time to debug it.

kuasha, 2018-05-28:
I am using the city environment (precompiled windows binary) on windows 10. I am trying to use Logitech G920 with following settings:

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "SimMode": "Car",
  "PhysXCar" : {
    "RC": {
      "RemoteControlID": 0
    }
  }
}
```
I see that the settings file is being picked up by the simulator but it still shows input mode to be keyboard. I can drive with arrow keys. I have downloaded the environment 2 days ago. Device manager shows the controller as Xbox Gaming Device.

hrishikeshtawade04, 2018-08-30:
I have downloaded the simple maze precompiled binary of airsim. I want to connect the G920 wheel to it. How to do it?

hrishikeshtawade04, 2018-09-06:
I have Logitech G920 steering wheel and tried connecting it to the blocks environment. I edited the settings.json file and now it looks like this.
{
"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
"SettingsVersion": 1.2,
"PhysXCar" : {
"RC": {
"RemoteControlID": 0
}
}

}

But when I stop and play the unreal engine I am still getting the control mode as keyboard. Is there something else to be done?

sytelus, 2018-09-24:
There had been change in settings.json schema. The PhysXCar should be under Vehicles section. Please see settings.md in docs folder.

hrishikeshtawade04, 2018-09-27:
So if I add the vehicles section in the settings.json file and connect my controller and open the blocks environment it will connect right?

hrishikeshtawade04, 2018-09-27:
These are my current settings. Are they proper?
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "ViewMode": "",
  "Vehicles": {
    "PhysXCar": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "RC": {
        "RemoteControlID": 0
      }	     
    }
  }
}

tfromb, 2019-02-22:
Hi, Im having the same issue. I am using a G920 and followed the instructions however I can't use the pedals to move the vehicle. The simulator picks up on the steering angles. What changes should be made to the setting.json file to enable the steering wheel to fully control the vehicle? Thank you

wintersoom, 2019-07-09:
> Hi, Im having the same issue. I am using a G920 and followed the instructions however I can't use the pedals to move the vehicle. The simulator picks up on the steering angles. What changes should be made to the setting.json file to enable the steering wheel to fully control the vehicle? Thank you

HI, do you solve the problem? i have the same problem now

wintersoom, 2019-07-09:
> I have Logitech G920 steering wheel and tried connecting it to the blocks environment. I edited the settings.json file and now it looks like this.
> {
> "SeeDocsAt": [...]

HI, do you solve the problem? i have the same problem now

closed 2021-02-05 by jonyMarino