microsoft/AirSim #975: Getting truth data from sim over mavros interface (mavlink packets)

realdealneil, 2018-04-16:
I have been searching the AirSim code looking for whether the truth information from AirSim is published over mavlink when running HITL simulation.  I can't seem to find where that might happen.  I thought that possibly it would be published using the MoCap messages in mavlink, but those don't appear to be coming through anywhere.  Can anyone tell me how this is already being done?  If it is not being done, can someone point me to where packets like this would be published?  I have not yet been able to determine where I might publish information like this (I'm looking for where HIL messages are published, and I haven't been able to figure it out yet).

realdealneil, 2018-04-16:
I have figured out where the HIL messages are sent...in MavLinkDroneController in the sendHILSensor and sendHILGps functions.  I am adding one for sending HIL truth.  Please let me know if this is being done already somewhere else.  Having truth published over mavlink would be very handy.

sytelus, 2018-04-18:
We are not publishing ground truth on MavLink. Are there any messages available to do this? Please feel free to contribute pull request.

realdealneil, 2018-04-18:
There is a Motion Capture pose packet that might work.  However, I want to receive it on a companion computer, not directly on the autopilot.  I also want to be able to send true velocities (inertial and/or body angular and linear velocities), and I don't know of a packet for that.  I may have to construct one.  Has anyone done this before?

sytelus, 2018-04-25:
I think Motion Capture messages will interfere with FCs like PX4. The best way I can I think of is not to try to use MavLink for this and instead you can just use simple RPC calls to communicate additional data you need.

closed 2018-05-08 by sytelus