microsoft/AirSim #977: Binaries are not working

Pierrolo, 2018-04-17:
Hi I downloaded the NH environment from your binaries, but it doesn't work. 
At the start, I choose multirotor ,so I click "No" and then it displays the following:

[image]

and then : 
[image]

I tried launching it on two different computers and I get the same error. I also tried to run the latest ZhangJiaJie, and the same error poped up. Those same error also happen when I try to select the car.
My belief is that, 5 days ago there were a few changes made, clockspeed for example, that made the build and the binaries not work. Then it got fixed for the build, but I don't think the binaries were updated...

Thanks

haider04, 2018-04-19:
I am having this same problem. But my city environment just worked fine. Having problems with NH and Africa binaries. 
Any solution so far?

Pierrolo, 2018-04-19:
Hi,
I just downloaded the city env, and got the same problem personnaly.

totht91, 2018-04-19:
prerelease is broken for us too.

allen9273, 2018-04-22:
City environment is fine in car mode, but drone mode is broken down.

QiyuZ, 2018-04-24:
The same question. However the in the previous version, NH can work

vinbo, 2018-06-10:
@sytelus I still met the similar problem in 1.1.10, with pop up box becoming" error at start up: error creating directiories, hr=5" , how can i solve this, please? 
[image]
and click "OK" and then
[image]

msb336, 2019-10-03:
as @vinbo 's issue is separate from the problem listed in this thread, let's track it in #1150 

The issue in this thread should be solved with the current binary releases.

closed 2019-10-03 by msb336