microsoft/AirSim #978: How to set up two or more different vehicles and connect them to Pixhawks via USB???

Dflybird, 2018-04-17:
I want to connect **two** Pixhawks to the AirSim.
Does the simulation supports two or more vehicles running in the environment ?
My settings.json:
```
{
  "SettingsVersion": 1.0,
  "SimMode": "Multirotor",
  "DefaultVehicleConfig": "PX4"
  "PX4_1": {
    "FirmwareName": "PX4",
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "SerialPort":"COM3",
    "ApiServerPort": 41451
  },
  "PX4_2": {
    "FirmwareName": "PX4",
    "SitlIp": "127.0.0.1",
    "SitlPort": 14557,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14561,
    "UseSerial": true,
    "SerialPort":"COM4",
    "ApiServerPort": 41452
  }
}
```
but the AirSim could only found one pixhawk. 
As the picture below...
[image]
It could only connect COM3,  but can't found COM4.
What should I do to connect both of my Pixhawks to the AirSim?????
PLEASE HELP ME!!! 
THANK YOU!

Isaac45, 2018-05-09:
@Dflybird, having the same problem here. Please contact me if you get the solution

jonyMarino, 2021-06-01:
Hi. Please take a look at [the docs](https://microsoft.github.io/AirSim/px4_multi_vehicle/)
Closed due to inactivity. Please, feel free to ask for it to be reopened if you wish to continue posting.

closed 2021-06-01 by jonyMarino