microsoft/AirSim #985: Yaw and move continuously

guillem74, 2018-04-18:
I am selecting three actions for my reinforcement learning project, one is going ahead, and the other ones are yaw left and right with different angles. Here you have the code:

```
def straight(self, duration, speed):
        pitch, roll, yaw  = self.getPitchRollYaw()
        vx = math.cos(yaw) * speed
        vy = math.sin(yaw) * speed
        self.moveByVelocityZ(vx, vy, self.z, duration, DrivetrainType.ForwardOnly)
        start = time.time()
        return start, duration
    
    def yaw_right(self, duration):
        self.rotateByYawRate(30, duration)
        start = time.time()
        return start, duration
    
    def yaw_left(self, duration):
        self.rotateByYawRate(-30, duration)
        start = time.time()
        return start, duration
```

Using this I go ahead, then rotate, then go ahead after the rotation, and so on ....
But what I want is the ability to rotate and go ahead without stopping, like continuous movements (so as when you are driving a car and you turn right or left).

moveByVelocity() or moveByVelocityZ() do not suit for me because I need to get front images, and when using them, I am moving basically North, West, South, East but the front camera is not facing the direction of the drone.

Pierrolo, 2018-04-18:
Hi,

I have the same problem, here is what I do, it might help you:
```
    def move_forward_Speed(self, duration = 0.5, speed_x = 0.5, speed_y = 0.5):
        pitch, roll, yaw  = self.getPitchRollYaw()
        vel = self.getVelocity()
        vx = math.cos(yaw) * speed_x - math.sin(yaw) * speed_y
        vy = math.sin(yaw) * speed_x + math.cos(yaw) * speed_y
        if speed_x <= 0.01:
            drivetrain = DrivetrainType.ForwardOnly
            yaw_mode = YawMode(is_rate= True, yaw_or_rate = 0)
        elif speed_x > 0.01:
            drivetrain = DrivetrainType.ForwardOnly
            yaw_mode = YawMode(is_rate= False, yaw_or_rate = 0)
        self.moveByVelocityZ(vx = (vx +vel.x_val)/2 ,
                             vy = (vy +vel.y_val)/2 , #do this to try and smooth the movement
                             z = self.z,
                             duration = duration + 5,
                             drivetrain = drivetrain,
                             yaw_mode = yaw_mode)
```

I am not saying that this works very well, but I do believe it's the place to start, by taking 

```
drivetrain = DrivetrainType.ForwardOnly
yaw_mode = YawMode(is_rate= False, yaw_or_rate = 0)
```

lovettchris, 2018-04-19:
If you want to fly someplace with the front of the drone pointing elsewhere then take a look at PythonClient\Orbit.py - this script contains an example of flying in a circle with the front of the drone facing the center of the circle, using DrivetrainType.MaxDegreeOfFreedom, YawMode(False, camera_heading).

RVBldr, 2018-11-21:
Internal VSTF 7435.

closed 2020-09-14 by jonyMarino