microsoft/AirSim #986: Steering wheel and pedals not working in city environment

haider04, 2018-04-18:
Hi,
I have made changes in settings.json for my input steering wheel (SimSteering) and pedal (Fanatec) USB controllers as below:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "UsageScenario":"ComputerVision",
  "PhysXCar" : {
    "RC": {
        "RemoteControlID": 0
    }
} 
}
The RC controller on USB is detected and control mode is also Wheel/Joystick, but still car is not moving as shown in picture. The other thing is that Accel is always 0.5, Break: 2.0, Steering: 1.25 and RPM are continuously changing without having any input. 

Why is it happening and how i could make my car to move? Should i make some further changes in code or settings or where?

Thank you for your help and time. 

Regards,
Haider

[image]

sytelus, 2018-04-25:
You have set mode to computer vision. Remove line with `"UsageScenario":"ComputerVision"` and then chose car when AirSim starts,

haider04, 2018-04-27:
Hi @sytelus ,
I just removed computer vision mode and tried connecting steering wheel and pedals (USB controllers) in the latest city environment binary. Now my steering wheel vibrates during simulation but rotates continuously to the left, pedals are still not working and the control values keep changing like the picture shown above and the car stands still.

Can you please recommend where should i start looking to change code to integrate my steering wheels and pedals, also in which and in how many files i have to change code or do something in settings.json file? 

I am using SimSteering2 force feedback system from Leobodnar.

I'll really appreciate your help in this regard. 

Regards,
Haider

sytelus, 2018-04-27:
This should be fixed now (only G920 wheels are supported). Duplicate of https://github.com/Microsoft/AirSim/issues/1024.

closed 2020-08-17 by jonyMarino