microsoft/AirSim #987: msgpackrpc.error.TransportError: Retry connection over the limit

husha1993, 2018-04-18:
System configuration : Ubuntu 16.04, 64 bit.
I have quoted the exact error log bellow. Any help or a hint would be appreciated.

python hello_drone.py    

WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
Traceback (most recent call last):
  File "hello_drone.py", line 14, in <module>
    client.confirmConnection()
  File "/cs/vml4/shah/AirSim/PythonClient/AirSimClient.py", line 220, in confirmConnection
    home = self.getHomeGeoPoint()
  File "/cs/vml4/shah/AirSim/PythonClient/AirSimClient.py", line 229, in getHomeGeoPoint
    return GeoPoint.from_msgpack(self.client.call('getHomeGeoPoint'))
  File "/cs/vml4/shah/anaconda2/envs/pytorch/lib/python3.6/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/cs/vml4/shah/anaconda2/envs/pytorch/lib/python3.6/site-packages/msgpackrpc/future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit

husha1993, 2018-04-18:
https://github.com/Microsoft/AirSim/blob/b6ee2e6045c5b19471d25353cc419d5592020409/PythonClient/AirSimClient.py#L2

Here the comment said "#install as admin: pip install msgpack-rpc-python"

sytelus, 2018-04-25:
Either you don't have AirSim running in UE or have issues in permissions to connect to server.

closed 2018-05-01 by sytelus

zenghaogithub, 2020-03-31:
> have AirSim running in UE or have issues in permissions to connect to server.

 how can I have AirSim running in UE or have issues in permissions to connect to server？Any help or a hint would be appreciated.

zenghaogithub, 2020-03-31:
> https://github.com/Microsoft/AirSim/blob/b6ee2e6045c5b19471d25353cc419d5592020409/PythonClient/AirSimClient.py#L2
> 
> Here the comment said "#install as admin: pip install msgpack-rpc-python"

i run the code on windows10，the error is the same as you. I have tryed to install  msgpack-rpc-python successfully, but it doesn't work. If you can provide something helpful, I would be appreciated.

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "E:\Program Files (x86)\pycharm\PyCharm 2019.1.3\helpers\pydev\_pydev_bundle\pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "E:\Program Files (x86)\pycharm\PyCharm 2019.1.3\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:/Users/Administrator/Desktop/强化学习/sqn_rand_v2.py", line 13, in <module>
    client.confirmConnection()
  File "E:\Program Files (x86)\anaconda3\envs\gym\lib\site-packages\airsim\client.py", line 54, in confirmConnection
    if self.ping():
  File "E:\Program Files (x86)\anaconda3\envs\gym\lib\site-packages\airsim\client.py", line 24, in ping
    return self.client.call('ping')
  File "E:\Program Files (x86)\anaconda3\envs\gym\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "E:\Program Files (x86)\anaconda3\envs\gym\lib\site-packages\msgpackrpc\future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit

wss020, 2020-05-28:
I have face the same problem. Have you solve this? @zenghaogithub

postureai, 2020-07-14:
Did you guys able to solve it? @wss020 @zenghaogithub

OceanYing, 2020-09-04:
I have faced exactly the same problem before. Maybe you should open your UE window and click the 'run'  before executing hello_car.py.

acen23, 2022-08-05:
I'm experiencing the exact same issue with my UE engine running the simulation already. I already installed msgpack-rpc-python, but am not sure how to ensure permissions to connect to server if that's the issue here. Any help would be appreciated!