microsoft/AirSim #997: Build Fail UE4 with code modification

shriram17, 2018-04-20:
Hey, 
I'm currently using AirSim plugin in my own UE4 project and modified a Show Message string in the existing AirSim code (SimHUD.cpp, Line 256). I also copied the .bat files from AirSim\Unreal\Environments\Blocks as mentioned in the [FAQ section](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md#is-there-easy-way-to-sync-code-in-my-unreal-project-with-code-in-airsim-repo).
On building the project through VS I encounter these errors

> \Plugins\AirSim\Source\SimHUD\SimHUD.cpp(35): error C2653: 'UAirBlueprintLib': is not a class or namespace name
\Plugins\AirSim\Source\SimHUD\SimHUD.cpp(35): error C3861: 'ShowMessage': identifier not found
\Plugins\AirSim\Source\SimHUD\SimHUD.cpp(56): error C2064: term does not evaluate to a function taking 0 arguments

Are there any addition steps to be performed before building the project through VS2017?
**Note : Same code modification compiled and worked in the Blocks project.**

sytelus, 2018-04-25:
Unfortunately we have limited bandwidth in debugging custom modifications. You should generate .sln file, load it up in VS and see where the error is occurring and why.

sytelus, 2018-04-29:
Just encountered these errors and traced back to IWYU mode in UE 4.18. It was just matter of including all needed UE headers. This is now fixed in this commit: https://github.com/Microsoft/AirSim/commit/7bddd5857791f9c164e8eba80c229f199c0babf8 .

closed 2018-04-30 by sytelus