microsoft/AirSim #998: source code or binaries-linux  of CityEnviron

husha1993, 2018-04-20:
Hi,

Will the source code or binaries-linux  of  CityEnviron be released?
Currently I only find the binaries-windows of CityEnviron here:
https://github.com/Microsoft/AirSim/releases

Thanks.

sytelus, 2018-04-25:
Yes, we are planning to release CityEnviron binaries after we complete upgrade to UE 4.18 - hopefully in couple of weeks.

Domi47, 2018-05-28:
Hi @sytelus, i need an open environment like this city neighborhood for linux and was thinking about buying it. Do you still plan to release it soon? this environment allows you to fly inside buildings too?
Thanks!

sytelus, 2018-10-17:
The City Environment binaries are now released for Linux!

https://github.com/Microsoft/AirSim/releases/tag/untagged-a395e93dd037d937537a

closed 2018-10-17 by sytelus

abinlee, 2018-10-17:
I got "Page not fount" when I click "https://github.com/Microsoft/AirSim/releases/tag/untagged-a395e93dd037d937537a".

micmarty, 2018-11-05:
Could someone refresh that link 🥇 ??? I'm starving here, haha

kang235, 2018-11-13:
I cannot open the link either. Could you please update it @sytelus? Thanks.