microsoft/AirSim #1001: Getting _ITERATOR_DEBUG_LEVEL errors during AirSim Windows Build

IAmSaad, 2018-04-21:
I am following the Build AirSim Windows guide [here](https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md) and I get the following errors when building the RPC module.

[image]

I saw stackoverflow ([this link](https://stackoverflow.com/questions/7668200/error-lnk2038-mismatch-detected-for-iterator-debug-level-value-0-doesnt)) and found some pointers on what may be causing it. This has got something to do with DEBUG and RELEASE versioning. I opened AirSim.sln and manually started adding the preprocessor flags `_HAS_ITERATOR_DEBUGGING=0;_ITERATOR_DEBUG_LEVEL=0;` to AirLib, AirlibUnitUnitTests, DroneServer etc and rebuilding them in Visual Studio but that didn't change the build errors.

sytelus, 2018-04-25:
You shouldn't be getting any such errors. The build command should work without error. Are you using VS2015 Update 3? Are you doing build command from VS x64 command prompt?

IAmSaad, 2018-04-25:
Yes I have a fresh install VS2015 Community Update 3 and I am doing it in the VS2015 x64 command prompt. If I use generic command prompt, it doesn't find msbuild.

sytelus, 2018-04-25:
There was a bug in DroneServer that got just fixed although I don't think its relevant to what you are reporting. For the latest person, I can do build without any errors. You might want to pull latest, make sure you don't have any custom changes and just run build command.

IAmSaad, 2018-04-27:
Still getting the same error on the latest pull of master. Are the subprojects like DroneServer, Examples, HelloCar, DroneShell etc are being released with preprocessor flag _ITERATOR_DEBUG_VALUE = 0 or debugging turned off? Because all subprojects build fail on my side with this 

> mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value
'2' in main.obj

sytelus, 2018-04-27:
Could you run clean.bat and regenerate solution. I'm not getting this error.

IAmSaad, 2018-04-30:
Thanks! It builds fine now and I have used VS2017 for the build process now.