microsoft/AirSim #1002: What is the wheel control shown in "AirSimCarManual.gif"

ralphpoon, 2018-04-22:
I would like to ask what is the wheel controller shown in "AirSim/docs/images/AirSimCarManual.gif" ?
Is it a game controller for Xbox or PS3? Thanks.

[image]

mitchellspryn, 2018-04-24:
Oh hey, that's my hand in that gif :P 

That is an old steering wheel that is no longer available (to my knowledge). We've gotten great results using [this one](https://www.walmart.com/ip/Thrustmaster-Ferrari-Red-Legend-Race-Wheel-PS3-PC/30824036?wmlspartner=wlpa&selectedSellerId=0&adid=22222222227021749969&wl0=&wl1=g&wl2=c&wl3=40342616792&wl4=pla-78302436992&wl5=9033315&wl6=&wl7=&wl8=&wl9=pla&wl10=8175035&wl11=online&wl12=30824036&wl13=&veh=sem)

sytelus, 2018-04-25:
You can use Logitech G920. See https://github.com/Microsoft/AirSim/issues/897#issuecomment-375550730

closed 2018-05-01 by sytelus