microsoft/AirSim #1004: has_collided not updated

niknil702, 2018-04-23:
When collision occurs the _has_collided_ sometimes remains set to **False** even though the values of the _impact_point_ vector is updated. The below shows two readings of collision info, first one before any collision and the second one after the first collision. 

`
<CollisionInfo> {   'has_collided': False,
    'impact_point': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'normal': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'object_id': -1,
    'object_name': '',
    'penetration_depth': 0.0,
    'position': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'time_stamp': 0}
reward -1.0591950962437693
total reward: -2.243125530113673
<CollisionInfo> {   'has_collided': False,
    'impact_point': <Vector3r> {   'x_val': 7.556523323059082,
    'y_val': -3.422746419906616,
    'z_val': -4.774749755859375},
    'normal': <Vector3r> {   'x_val': -2.086162567138672e-06,
    'y_val': 0.9999998807907104,
    'z_val': 0.0},
    'object_id': 181,
    'object_name': 'wall3',
    'penetration_depth': 9.999999974752427e-07,
    'position': <Vector3r> {   'x_val': 7.556523323059082,
    'y_val': -3.422746419906616,
    'z_val': -4.774749755859375},
    'time_stamp': 1524477282691448320}
`

husha1993, 2018-04-23:
I think maybe when there is no collision detected ('has_collided': False), then the data of 'impact_point', 'normal', 'penetration_depth' is random generated and they are not true data .

Hoshea7, 2019-08-01:
Hi @niknil702 , 

I meet the same issue when I use ('has_collided'), which cannot return False when meet the collision immediately, till now have you sorted out this issue? or any solutions you take to collect the collision information?

niknil702, 2019-08-06:
> Hi @niknil702 ,
> 
> I meet the same issue when I use ('has_collided'), which cannot return False when meet the collision immediately, till now have you sorted out this issue? or any solutions you take to collect the collision information?

Yes, my workaround was to look at the timestamp instead. I kept track of the last timestamp and considered it a collision if the timestamp changed. Timestamp field seems reliable.

riyadfebrian, 2021-12-21:
for me, `impact_point` and `position` are always changing, even without any collision. 
but for `normal` it always change when my agent hit the obstacle. 

take a look at my answer at this issue #1986

zimmy87, 2021-12-22:
I did some testing locally with the simGetCollisionInfo() and I do see collision_info.has_collided get set to true, but it gets reset to false in PawnSimApi::setPoseInternal, so it looks like the true value can only be read in the clock tick immediately after a collision: if any more time elapses between a collsion and a call to simGetCollisionInfo(), you will read collision_info.has_collided after it has reset to false. It's possible this behavior can be changed to only reset collision_info.has_collided after a call to simGetCollisionInfo(), but some testing will need to be done to see if this is a safe change because this variable is being checked elsewhere in PawnSimApi::canTeleportWhileMove(), so there may some unintended consequences to making a change like this.

PR #4237 (2022-01-11): keep CollisionInfo.has_collided flag set until simGetCollisionInfo called

Fixes: #1004    

## About

This adds a new method to PawnSimApi called getCollisionInfoAndReset() which both returns collision info as well as reset the has_collided flag. The old behavior for resetting CollisionInfo.has_collided after every clock tick has been removed and simGetCollisionInfo now calls getCollisionInfoAndReset(), which means this method will return whether any collision has occurred since last call to simGetCollisionInfo instead of only within the last clock tick.

## How Has This Been Tested?

Collisi [... cut]
closed 2022-01-11 by jonyMarino