microsoft/AirSim #1006: no moving vehicles and humans

husha1993, 2018-04-23:
Hi,

I  tried the most recent released windows-binaries of CityEnviron:
https://github.com/Microsoft/AirSim/releases

But I can't find any moving vehicles and humans within.
Does the current version contain moving vehicles and humans? 

Thanks.

allen9273, 2018-04-24:
Same problem. There are no moving vehicles and people in v1.1.9, but you can find them in v1.1.8.

husha1993, 2018-04-24:
I can find the moving vehicles in v1.1.8 but can't find the moving people in v.1.1.8...

allen9273, 2018-04-25:
I found people in v1.1.8 around the houses.
[image]

sytelus, 2018-04-25:
Could you please try latest release. There are bunch of pedesterians in there: https://github.com/Microsoft/AirSim/releases/tag/v1.1.10

closed 2018-05-08 by sytelus