microsoft/AirSim #1007: Don't show car info on screen

NextSim, 2018-04-24:
How  can I turn off the data shown on the screen (accel, brake, control mode, etc)?

I looked through the settings and didn't see anything

NextSim, 2018-04-24:
I tried adding that and it didn't affect it.  I'll try using the console, but I need to be able to have it launch that was or programatically remove it.

Here is my setting.json

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "SimMode": "Car",
  "RecordUIVisible": false,
  "LogMessagesVisible": false
}

```

NextSim, 2018-06-20:
Is this something that is set when the project is packaged?

I have tried on a few scenarios, on City, Africa and Neighborhood the log shows, regardless of what is in the settings file.

I created a project from landscape mountains, and this project, as well as the blocks project, never shows the info, regardless of what is in the settings file.

rajat2004, 2020-08-17:
@jonyMarino Actually this problem still exists in master, so this issue can be reopened I think

jonyMarino, 2020-08-17:
@rajat2004 Thank you. I thought it was a duplicate of  #1583 (closed)

rajat2004, 2020-08-17:
Yeah, that issue seems to have been closed to keep discussion here. Will try to look into this and maybe open a PR
Thanks for all the hard work you and the other maintainers have been putting in to clean up the issues!

PR #2946 (2020-09-01): Disable LogMessages if set to false
Settings - 
```
{
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "RecordUIVisible": false,
  "LogMessagesVisible": false
}
```

[image]

Closes #1007

If OnScreenMessages aren't cleared, then the settings log message appears since it's being done before others, remains for 60sec.
Thanks to #1583 for how to clear the debug messages
closed 2020-09-01 by madratman