microsoft/AirSim #1009: DQNcar in PythonClient is not work (will not converge)

QiyuZ, 2018-04-25:
There is an example in PythonClient which applies DQN to run the car. However, I ran it for a long time and found it will not converge. Does anyone success?

mitchellspryn, 2018-04-25:
Not sure about the tutorial on this page, but you may try ours [here](https://github.com/Microsoft/AutonomousDrivingCookbook/tree/master/DistributedRL).

zimmy87, 2021-01-15:
Hi @QiyuZ, thank you for reporting this issue.  This script was recently removed in #3215 and replaced with a new script at \AirSim\PythonClient\reinforcement_learning\dqn_car.py that uses the new airgym module. I tested the new script locally and it runs fine for me. Can you retest with the new script and confirm whether you are still seeing this issue or not?

closed 2021-01-15 by zimmy87