microsoft/AirSim #1010: building issues on airsim

z2140684, 2018-04-25:
when i run ./buidl.sh    it occurs,looking for help

[ 98%] Building CXX object DroneServer/CMakeFiles/DroneServer.dir/home/ubuntu/airsimenv/AirSim/DroneServer/main.cpp.o
/home/ubuntu/airsimenv/AirSim/DroneServer/main.cpp:100:27: error: variable type
      'DroneServerSimModeApi' is an abstract class
    DroneServerSimModeApi simmode_api(&api);
                          ^
/home/ubuntu/airsimenv/AirSim/AirLib/include/api/SimModeAPiBase.hpp:16:18: note: 
      unimplemented pure virtual method 'isPaused' in 'DroneServerSimModeApi'
    virtual bool isPaused() const = 0;
                 ^
/home/ubuntu/airsimenv/AirSim/AirLib/include/api/SimModeAPiBase.hpp:17:18: note: 
      unimplemented pure virtual method 'pause' in 'DroneServerSimModeApi'
    virtual void pause(bool is_paused) = 0;
                 ^
/home/ubuntu/airsimenv/AirSim/AirLib/include/api/SimModeAPiBase.hpp:18:18: note: 
      unimplemented pure virtual method 'continueForTime' in
      'DroneServerSimModeApi'
    virtual void continueForTime(double seconds) = 0;
                 ^
1 error generated.
DroneServer/CMakeFiles/DroneServer.dir/build.make:62: recipe for target 'DroneServer/CMakeFiles/DroneServer.dir/home/ubuntu/airsimenv/AirSim/DroneServer/main.cpp.o' failed
make[2]: *** [DroneServer/CMakeFiles/DroneServer.dir/home/ubuntu/airsimenv/AirSim/DroneServer/main.cpp.o] Error 1
CMakeFiles/Makefile2:556: recipe for target 'DroneServer/CMakeFiles/DroneServer.dir/all' failed
make[1]: *** [DroneServer/CMakeFiles/DroneServer.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

nelsondmmg, 2018-04-25:
The same error happened to me. Just to clarify, after a first mistake. First, the includes of the file RcpLibServerBase.hpp were:

`#include "common/Common.hpp"
#include "api/ApiServerBase.hpp"
#include "api/SimModeAPiBase.hpp"`

This I corrected to:

`#include "common/Common.hpp"
#include "ApiServerBase.hpp"
#include "SimModeApiBase.hpp"`

But after the execution of setup.sh, this error pop up:

/home/vedecom/simulation/AirSim/DroneServer/main.cpp:100:27: error: variable type 'DroneServerSimModeApi' is an abstract class
    DroneServerSimModeApi simmode_api(&api);
                          ^
/home/vedecom/simulation/AirSim/AirLib/include/api/SimModeApiBase.hpp:16:18: note: unimplemented pure virtual method 'isPaused' in 'DroneServerSimModeApi'
    virtual bool isPaused() const = 0;
                 ^
/home/vedecom/simulation/AirSim/AirLib/include/api/SimModeApiBase.hpp:17:18: note: unimplemented pure virtual method 'pause' in 'DroneServerSimModeApi'
    virtual void pause(bool is_paused) = 0;
                 ^
/home/vedecom/simulation/AirSim/AirLib/include/api/SimModeApiBase.hpp:18:18: note: unimplemented pure virtual method 'continueForTime' in 'DroneServerSimModeApi'
    virtual void continueForTime(double seconds) = 0;

sytelus, 2018-04-25:
Just fixed this. Please try the latest. Thanks for reporting.

closed 2018-04-25 by sytelus