microsoft/AirSim #1011: Unreal Editor crashing when opening a new C++ project

dileep-krishnan, 2018-04-25:
I am trying to run AirSim in Linux. When I try to open Unreal Editor with a blank C++ project, it closes. Below is the log.

```
Starting /home/dileep/AirSim/UnrealEngine/Engine/Binaries/Linux/UE4Editor...
Increasing per-process limit of core file size to infinity.
LogInit: Display: Running engine without a game
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(716223097) SRandInit(716223097).
LogTaskGraph: Started task graph with 5 named threads and 14 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.049361
LogPluginManager: Mounting plugin LevelSequenceEditor
LogPluginManager: Mounting plugin MatineeToLevelSequence
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin GoogleCloudMessaging
LogPluginManager: Mounting plugin AndroidDeviceProfileSelector
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin AppleMoviePlayer
LogPluginManager: Mounting plugin IOSDeviceProfileSelector
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin AndroidMoviePlayer
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin MediaPlayerEditor
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin OnlineSubsystemIOS
LogPluginManager: Mounting plugin OnlineSubsystemGooglePlay
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin SubversionSourceControl
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin HTML5Networking
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin PluginBrowser
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin SpeedTreeImporter
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin GameplayTagsEditor
LogPluginManager: Mounting plugin MobileLauncherProfileWizard
LogInit: Using libcurl 7.48.0-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
[... 838 lines cut ...]
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module LinuxDeviceProfileSelector (40)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module ExampleDeviceProfileSelector (38)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module SteamVR (36)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module AnimationModifiers (34)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module PropertyEditor (33)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module AudioEditor (30)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module TextureCompressor (28)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module ShaderCore (26)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module Landscape (24)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (22)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module OpenGLDrv (20)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (18)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module Renderer (16)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module Engine (14)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module CoreUObject (12)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module NetworkFile (10)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module CookedIterativeFile (8)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module StreamingFile (6)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module SandboxFile (4)
[2018.04.25-11.14.57:979][351]LogModuleManager: Shutting down and abandoning module PakFile (2)
[2018.04.25-11.14.57:980][351]LogExit: Exiting.
[2018.04.25-11.14.57:980][351]LogInit: Tearing down SDL.
/home/dileep/AirSim/UnrealEngine/Engine/Binaries/Linux/UE4Editor exited with code 0
```
Sorry for the long post.

sytelus, 2018-04-25:
There is no error messages in this logs. Please look in to ./Saved/Logs/ folder to see crash logs. Also, try to see what error is and post only relevant part of logs with your own observations and details.

closed 2018-05-08 by sytelus

basavaraj1995, 2019-11-21:
Okay once project is created, go to project directory and run command in terminal "make"
after complication double click on unreal project and it will pop up as rebuild the project  give "yes" and wait for some 60 sec and done.