microsoft/AirSim #1014: Getting distances without recording

foromer4, 2018-04-25:
I would like to know the distance to other vehicles visible in the scenario. I prefer not to use depth maps as suggested here: #887 because: 
a.Adding any  sort of recorder slows down the simulation FPS considerably. 
b (if possible) I would like to get the distance per actor so that I can track its velocity.
I know Unreal server keeps this data, but I couldn't see how to get it from the APIs.
thanks,
Omer

sytelus, 2018-04-25:
May be you can use [simGetObjectPose](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#common-apis)?

foromer4, 2018-04-26:
thanks, I saw this option, but how would I know which actor id to give it?

sytelus, 2018-04-26:
Right now this requires loading project in Unreal Editor and figure out names of actors/meshes. As we only distribute binaries this is not possible to do for our environments. May be we can add API to enumerate available actors in the environment. Please feel free to contribute this feature :).

msb336, 2019-10-04:
There is now an API called `simListSceneObjects`, which will return a list of all Actors in the scene.

closed 2019-10-04 by msb336