microsoft/AirSim #1015: Api params

NextSim, 2018-04-25:
Is there any documentation on the params that are to be sent with the api command?

I know some of the api is mentioned here:

https://github.com/Microsoft/AirSim/blob/master/docs/apis.md

but it seems that using the api for requests like reset and getCarState are not as simple as sending the api command.

sytelus, 2018-04-25:
Not sure what you mean by "sending" API command. Both of API calls are simply function calls in Python.

NextSim, 2018-04-26:
Sending to AirSim over the tcp connection.  So data has to be serialized into the msgpack-rpc format and sent over the tcp connection to AirSim

[image]

closed 2018-05-02 by NextSim