microsoft/AirSim #1016: simGetImages Depth Planner type doesn't work on Windows for high resolution

olivia-skydio, 2018-04-26:
So I am trying to run Airsim in CV mode to collect data and want to get 2000x2000 resolution images. I know this works as I've done it in Linux and have been successfully able to collect thousands of images. 

I'm trying to replicate this process on Windows and have been unsuccessful, specifically when I have an image request where pixel_as_float is set to True and my resolution is higher than 600 (works fine if pixel_as_float is false). Airsim runs extremely slowly in the simGetImages() function, and sometimes when it does finish running simGetImages I get an "IOError [Errno 0]" when trying to run a print statement following it. 

Any idea what  could fix this besides reducing the resolution?  Thanks!

sytelus, 2018-04-26:
Just tried this out... 

- Used Neighbourhood environment binaries from latest release
- Used below setting.json
- Press R key to start and then stop recording
- I have TitanX GPU and SSD

I got 1.7 files/sec and no errors.

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "UsageScenario": "ComputerVision",
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraID": 0, "ImageType": 1, "PixelsAsFloat": true, "Compress": false }
    ]
  },
  "CaptureSettings": [
    {
      "ImageType": 1,
      "Width": 2000,
      "Height": 2000
	}
   ]
}
```

olivia-skydio, 2018-04-26:
Right, I agree I know this works since I've done it on Linux. Do you know any reason why this might be an issue? Or anything that could potentially cause this?

olivia-skydio, 2018-05-08:
So after some investigation, there isn't an issue with serializing the data (happens within milliseconds) with msgpack, the issue is with rpc server side, where it gets stuck waiting/stalling  (on something I haven't determined yet) when sending the data.

sytelus, 2018-05-08:
That's good to know... Is there any specific method? If it is possible to repro this in minimal code setting, we can try take a look.

sytelus, 2018-10-17:
If this is still a problem, please file a new issue with code, settings and clear steps to reproduce the issue.

closed 2018-10-17 by sytelus

Arnav14M, 2018-12-18:
I am having the same issue as olivia.
If your resolution is too high(i tried with 1920 x 1080), the airsim.ImageRequest(0, airsim.ImageType.DepthVis, True) where pixel_as_float is set to True will cause the code to freeze.