microsoft/AirSim #1020: cant open Blocks.uproject

z2140684, 2018-04-26:
when i finished building of UN4 and AirSim and try to load Blocks.uproject to test 
i was promted : 
                                      Missing Blocks Modules
       The following module are missing or built with a different engine version:
       libUE4Editor-Blocks.so
       libUE4Editor-AirSim.so
       Would you like to rebuild then now?
when i choose No ,nothing happend,
when i choose yes,i was prompted:
                       Error
       Blocks could not be compiled.Try rebuilding from source manually
i try to run clean.sh but get same situation.hope anyone can help me,thanks

elgeom, 2018-04-26:
I have the same issue with latest pull.
Ubuntu 16.04

sytelus, 2018-04-26:
Just fixed it: https://github.com/Microsoft/AirSim/issues/1023

madratman, 2019-01-22:
@z2140684 this is pretty late but probably useful for @Officepot, please select "in-place conversion" which is under the "more" options when unreal throws that error. It's in step 4 of the readme. 
@Officepot can you try this and re-open if it still doesn't work. We also have the blocks binary under the linux releases tab

closed 2019-01-22 by madratman

Neabfi, 2020-01-08:
How can I do the "in-place conversion” on a headless server?

```
LogInit: Warning: Incompatible or missing module: libUE4Editor-Blocks.so
LogInit: Warning: Incompatible or missing module: libUE4Editor-AirSim.so
LogInit: Initializing SDL.
LogInit: Using SDL_WINDOW_OPENGL
LogInit: Initialized SDL 2.0.5 revision: 11184 (hg-11184:04063928c4a8 with EpicExtensions) (compiled against 2.0.5)
LogInit: Display metrics:
LogInit:   PrimaryDisplayWidth: 1024
LogInit:   PrimaryDisplayHeight: 768
LogInit:   PrimaryDisplayWorkAreaRect:
LogInit:     Left=0, Top=0, Right=1024, Bottom=768
LogInit:   VirtualDisplayRect:
LogInit:     Left=0, Top=0, Right=1024, Bottom=768
LogInit:   TitleSafePaddingSize: X=0.000 Y=0.000
LogInit:   ActionSafePaddingSize: X=0.000 Y=0.000
LogInit:   Number of monitors: 1
LogInit:     Monitor 0
LogInit:       Name: 0
LogInit:       ID: display0
LogInit:       NativeWidth: 1024
LogInit:       NativeHeight: 768
LogInit:       bIsPrimary: true
LogLinux: Warning: MessageBox: The following modules are missing or built with a different engine version:

libUE4Editor-Blocks.so
libUE4Editor-AirSim.so

Would you like to rebuild them now?: Missing Blocks Modules: No is implied.
LogExit: Preparing to exit.
LogModuleManager: Shutting down and abandoning module AnimationModifiers (34)
LogModuleManager: Shutting down and abandoning module PropertyEditor (33)
LogModuleManager: Shutting down and abandoning module AudioEditor (30)
LogModuleManager: Shutting down and abandoning module TextureCompressor (28)
LogModuleManager: Shutting down and abandoning module ShaderCore (26)
LogModuleManager: Shutting down and abandoning module Landscape (24)
LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (22)
LogModuleManager: Shutting down and abandoning module OpenGLDrv (20)
LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (18)
LogModuleManager: Shutting down and abandoning module Renderer (16)
LogModuleManager: Shutting down and abandoning module Engine (14)
LogModuleManager: Shutting down and abandoning module CoreUObject (12)
LogModuleManager: Shutting down and abandoning module NetworkFile (10)
LogModuleManager: Shutting down and abandoning module CookedIterativeFile (8)
LogModuleManager: Shutting down and abandoning module StreamingFile (6)
LogModuleManager: Shutting down and abandoning module SandboxFile (4)
LogModuleManager: Shutting down and abandoning module PakFile (2)
LogExit: Exiting.
LogInit: Tearing down SDL.
Exiting abnormally (error code: 1)
```

tanyaapaul, 2022-10-05:
Did anyone managed to solve this issue? I currently have the same problem and not sure how to fixe it

ssaigarimella, 2024-11-05:
@tanyaspaul @Neabfi same issue, were you able to solve?