microsoft/AirSim #1023: Error compiling Blocks.uproject on Ubuntu

nelsondmmg, 2018-04-26:
I am having some issues trying to open the Blocks project, to test the AirSim framework. I followed the instructions in unreal_blocks.md for linux just until 5. After the in-place conversion the UE4 prompt a window saying that it is impossible to compile the project due to the following error:

`ERROR: UBT ERROR: Failed to produce item: /home/vedecom/simulation/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so`

I know that this is a generic error, but it is the only error message that I found in the window. I followed all the steps in the setup instructions for Linux.

elgeom, 2018-04-26:
I have the same three errors trying to compile libUE4Editor-Blocks.so and libUE4Editor-AirSim.so

JordanWang0223, 2018-04-26:
I have the same errors ... anyone fixed this?

nelsondmmg, 2018-04-26:
I fixed all problems (at least I think I did). For the error with the note just go to the file and remove the std::move cited. For all the other problems it is easier to copy the following files into SimJoyStick.h and .cpp:

**SimJoyStick.h**:

#pragma once

#include <cstdint>
#include <memory>
#include <string>
#include <stdint.h>
#include <limits>

class SimJoyStick
{
public:
    struct AxisMap {
        enum class AxisType : int {
            Auto = 0, LeftX, LeftY, LeftZ, RightX, RightY, RightZ
        };
        enum class AxisDirection : int {
            Auto = 0, Normal, Reverse
        };

        AxisType rc_axis = AxisType::Auto;
        float min_val = -1000, max_val = 1000;
        AxisDirection direction = AxisDirection::Auto;
    };
    struct AxisMaps {
        AxisMap left_x, left_y, left_z, right_x, right_y, right_z;
    } axis_maps;

    struct State {
        float left_x, left_y, left_z;
        float right_x, right_y, right_z;

        float slider0, slider1;
        float pov0, pov1, pov2, pov3;

        uint16_t buttons;
        bool is_initialized = false;
        bool is_valid = false;

        std::string message;
        std::string pid_vid;
        unsigned long connection_error_code = std::numeric_limits<unsigned long>::max();
    };

    void getJoyStickState(unsigned int index, State& state);
    // strength ranges from -1 to 1
    void setAutoCenter(unsigned int index, double strength);

    // strength ranges from 0 to 1
    void setWheelRumble(unsigned int index, double strength);
    
    SimJoyStick();
    ~SimJoyStick();    //required for pimpl
private:
    static bool initialized_success_;
    //bool api_control_enabled_ = false;

    struct impl;
    std::unique_ptr<impl> pimpl_;
};

**SimJoyStick.cpp**:

#include "SimJoyStick.h"

#if defined _WIN32 || defined _WIN64

[... 323 lines cut ...]
#endif

SimJoyStick::SimJoyStick()
{
    pimpl_.reset(new impl());
}
SimJoyStick::~SimJoyStick()
{
    //required for pimpl
}

void SimJoyStick::getJoyStickState(unsigned int index, SimJoyStick::State& state)
{
    pimpl_->getJoyStickState(index, state, axis_maps);
}

void SimJoyStick::setAutoCenter(unsigned int index, double strength)
{
    pimpl_->setAutoCenter(index, strength);
}

void SimJoyStick::setWheelRumble(unsigned int index, double strength)
{
    pimpl_->setWheelRumble(index, strength);
}

sytelus, 2018-04-26:
All of above is fixed now. Please try out latest version. Thanks for reporting this issue.

nelsondmmg, 2018-04-27:
With a new clone of the code, in the ./build.sh execution the following error occurs:

CMakeFiles/HelloDrone.dir/home/vedecom/simulation/AirSim/HelloDrone/main.cpp.o: In function main':
/home/vedecom/simulation/AirSim/HelloDrone/main.cpp:63: undefined reference to msr::airlib::MultirotorRpcLibClient::armDisarm(bool)
/home/vedecom/simulation/AirSim/HelloDrone/main.cpp:103: undefined reference to msr::airlib::MultirotorRpcLibClient::armDisarm(bool)
clang: **error**: linker command failed with exit code 1 (use -v to see invocation)

elgeom, 2018-04-27:
Now it is compiling on Linux, thanks