microsoft/AirSim #1024: Car behaves abnormally in newly released City environment when connected to steering wheel.

BrianN92, 2018-04-26:
I tried the v1.1.10 City map today. The vehicle could be controlled by keyboard perfectly. However, when I used the steering wheel, wired things happened: 
[image]

The steering angle was always 6.428572, Gear was R, Throttle was -1.
I used the x360ce to map the steering wheel, and put it in directory: "CityEnviron\CityEnviron\Binaries\Win64". I also put it in another directory: "AirSimNH\AirSimNH\Binaries\Win64" which makes me to use the steering wheel to control the vehicle in Neighborhood map.

I also tried remove the x360ce, which did not help.

sytelus, 2018-04-26:
x360ce requires custom config for your specific hardware. I f you are using Logitech G920 or equivalent, just remove x360ce stuff and add this in settings:

```
"PhysXCar" : {
  "RC": { "RemoteControlID": 0 }
}
```

BrianN92, 2018-04-26:
I just tried this way, which did not help either. However, it was different from before. The steering value was"-1.25" at first, and then I press the acceleration panel, I found that the steering value changed with the degree of force I press to the panel. the Throttle value was always 0.5

sytelus, 2018-04-26:
Which steering wheel are you using?

BrianN92, 2018-04-26:
Sorry for not clarifying. I am using [Thrustmaster](https://www.walmart.com/ip/Thrustmaster-Ferrari-Red-Legend-Race-Wheel-PS3-PC/30824036?wmlspartner=wlpa&selectedSellerId=0&adid=22222222227021749969&wl0=&wl1=g&wl2=c&wl3=40342616792&wl4=pla-78302436992&wl5=9033315&wl6=&wl7=&wl8=&wl9=pla&wl10=8175035&wl11=online&wl12=30824036&wl13=&veh=sem)

sytelus, 2018-04-26:
Thrustmaster support was very recently added by @kissslorinc (see https://github.com/Microsoft/AirSim/pull/979). I personally haven't tested it. May be @kissslorinc can help?

kissslorinc, 2018-04-26:
You definitely won't need x360ce if you try to use the native Thrustmaster support. You probably should remove x360ce altogether from the project folder to avoid conflicts.
I tested with a different wheel than yours (Thrustmaster T500RS), it is possible that it has a different axis layout than the Ferrari Red you're using. I'm trying to think of a way to test your wheel's axis layout to integrate into the system.
@sytelus Is the Thrustmaster support already compiled into the other environment binaries? I remember something similar when i tried to use my Thrustmaster wheel first (before changing the control axes and opening the pull request)

sytelus, 2018-04-27:
Yes, I created new release binaries that is in 1.1.10 now so it should be there.

BrianN92, 2018-04-27:
Hmmm, I will try to remove x360ce from the project folder first. Should I download the 1.1.10 again? Or maybe I need to modify the CarPawn.cpp?

haider04, 2018-04-27:
Jut to add a comment. 

I downloaded latest city environment binaries and now at least my steering wheel has some vibrations effect (in previous versions, it was not). But the steering wheel rotates itself continuously towards left and the values of RPM keep changing. Accelerator and brakes have no effect so far. 

Can some one suggest where should i do changes in code or settings.json file to integrate my steering wheel and pedals?

P.S: I am using SimSteering2 force feedback system from Leobodnar.

Regards,
Haider

kissslorinc, 2018-04-27:
It's probably the autocenter that's rotating your wheel to the left. If the axes aren't properly mapped, it might detect that your wheel is rotated to the right, and therefore wants to recenter it.
Could you tell me if the wheel is rotating when the car is stopped?
Also, if you want to map your own wheel's axes, you can do that in the CarPawn.cpp, in the updateCarControls() function. You might want to add a condition to the existing ones on your wheel's vendor id (you can get it from the _joystick_state variable).

kissslorinc, 2018-04-27:
@BrianN92 is your problem solved with the new release?

haider04, 2018-04-27:
Thanks @kissslorinc for your comment. 

I just checked the wheel direction and its rotating to the right. But the steering wheel is continuously steering it to the left with continuously changing RPMs. 

I have posted a picture of this as well in a separate issue #986 .

You can have a better idea by looking at the control values in the picture i have posted in that issue. If you can comment there and suggest me something, that would be great. Thank you!

BrianN92, 2018-04-27:
@kissslorinc No, it does not work. Same problems. I also removed all x360ce files.

sytelus, 2018-04-27:
I just made a change to limit force feedback/auto center to thrustmaster device. You can get latest try it out. Thank you for your help in testing this change.

BrianN92, 2018-04-28:
Hi @sytelus @kissslorinc , I downloaded the latest version of City environment binaries and Airsim v1.1.10. However, the problem was still the same. I have no idea what to do now. 

Besides, I tried to build the environment by myself, but I could not find the city map in Unreal.

closed 2022-04-28 by jonyMarino