microsoft/AirSim #1025: Error when reset in the binary City Environment v1.1.10

rohjunha, 2018-04-26:
I just ran City environment binary (v1.1.10) on Windows 10 with a settings.json below:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "LocalHostIp": "***.***.***.***",
  "RpcEnabled": true,
  "SimMode": "Car",
  "ViewMode": "Fpv",
  "EngineSound": false,
  "ClockSpeed": 2,
  "UsageScenario": "ComputerVision"
}.
Either with or without "ComputerVision" mode, a fatal error occurred when I try to do reset.
Here are a screenshot and an error message.

[image]

Fatal error!

Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0x00000000

0x000000001B4FC2F1
0x000000004F3F3F10
0x0000000024AFDD00
0x000000004EBD3B20
0x000000003850A6C0
0x000000003850A6C0

sytelus, 2018-04-26:
You are using this wrong. Please do not use FPV and car mode with ComputerVision mode. If you want ComputerVision mode, select multirotor. We are planning to revamps ComputerVision mode in near future to eliminate this confusion.

sytelus, 2018-04-27:
Just fixed this. Thanks for reporting!

closed 2018-04-30 by sytelus