microsoft/AirSim #1026: How can I enable the detection of collision with ground?

Amigoshan, 2018-04-27:
I notice that the collision with ground won't be return by getCollisionInfo(). 
When the drone hits the ground, there would be some message show on the screen. But the collision count remains 0. 
[image]

I'm wondering how can I get this information passed through the getCollisionInfo interface. 

I also had a quick look at the code. I found there are two different collision information, one is CollisionInfo the other one is CollisionResponseInfo. I'm not sure about the difference.

Amigoshan, 2018-04-27:
Sorry, my mistake. I found the api do returns collision with ground. It also detects collision with tree trunk. But the drone won't interact with tree leaves and branches. In the simulation, the drone will just go through the leaves and branches without any resistance. Is this a setting up problem of unreal environment?

sytelus, 2018-04-27:
Yes, I think there is issue with setting collison boxes around them.

jonyMarino, 2020-09-10:
May be related to #1004?

jonyMarino, 2021-08-19:
Answer: https://github.com/microsoft/AirSim/issues/3855#issuecomment-874984693
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem.
Thanks!

closed 2021-08-19 by jonyMarino