microsoft/AirSim #1027: NULL in g_pWheelRumbleHandle

rohjunha, 2018-04-27:
I am suffering from nullptr error of `g_pWheelRumbleHandle` in `DirectInputJoyStick.cpp` after pulling from the master branch. I guess the commit 5dd876c477f1ef49cab047e82592f61b8ef89306 might cause the problem for me. I am currently using G29 and my code was working well before pulling.

sytelus, 2018-04-27:
This might be from Thrustmaster support added by @kissslorinc (see #979). May be @kissslorinc can help?

sytelus, 2018-04-27:
Just checked-in potential fix. You can try out latest version and let us know if that worked.

kissslorinc, 2018-04-27:
thanks @sytelus for the hotfix!

rohjunha, 2018-04-27:
I guess this part should be modified as well: https://github.com/Microsoft/AirSim/blob/126900e52a54bd6ee1cc36476dc23fb22c23051b/Unreal/Plugins/AirSim/Source/SimJoyStick/DirectInputJoyStick.cpp#L83-L84
```
if (g_pAutoCenterHandle) {
			g_pAutoCenterHandle->SetParameters(&g_sAutoCenterConfig, DIEP_DIRECTION |
				DIEP_TYPESPECIFICPARAMS | DIEP_START);
		}
```

sytelus, 2018-04-28:
Agreed and done!

PS: if you get latest, we have now also upgraded to UE 4.18 and VS 2017.

rohjunha, 2018-04-30:
Thanks for the update. I will close this issue.

closed 2018-04-30 by rohjunha