microsoft/AirSim #1029: How to use the distance sensor?

galgyeony, 2018-04-27:
I am very happy to see the new "distance" sensor but don't know how to use it.

As far as I know, the distance sensor internally uses getRayLength(), which is actually implemented by UE4.
However, I do not know how to acquire the result (i.e., to log or display the distance value).

Is there an API or another way to do so?

Thank you very much for your support.

sytelus, 2018-04-27:
There is no Python API yet. We are planning to expose this soon.

galgyeony, 2018-04-30:
I know that there are some GPS related sensor APIs (e.g., getGPSLocation) and IMU related ones (e.g., get*Kinematics).
But, for me, it is still difficult to find any API for magnetometor and barometer.

Is there any API for magnetometor and barometer? If not, are you planning to expose them?

Thank you!

dileep-krishnan, 2018-05-25:
Hi @sytelus ,
Any update on the above feature?
If not, can you point me to an existing API as example so that I can build make the API myself.

Thanks a lot!!

paulerikf, 2018-09-06:
@dileep-krishnan Did you figure this out?  Alternatively, does anyone have an example?

dahewujiang, 2018-11-09:
hi, @sytelus is there anything updated for distance sensor ? I cannot  find docs for how to use this sensor. You know that it's necessary to mount sensors (e.g.,ultrasonic sensor,millimeter-wave radar) on cars  if  we want to test our driverless car. i think that we need more sensors model (ultrasonic sensor and millimeter-wave radar are best ).

stavBodik, 2019-04-21:
@sytelus @paulerikf @dileep-krishnan 

for C++ users

 I also could not find anyway to get Barometer pressure...but :

Please look at BarometerSimple.hpp

Function : getOutputInternal(...)

 auto pressure = EarthUtils::getStandardPressure(altitude);

. . .

madratman, 2019-04-30:
I just added APIs for Barometer, IMU, GPS, Magnetometer, Distance Sensor in https://github.com/Microsoft/AirSim/pull/1920 

See the updated docs https://github.com/Microsoft/AirSim/blob/master/docs/sensors.md
and the API here https://github.com/Microsoft/AirSim/blob/master/docs/sensors.md#sensor-apis

madratman, 2019-04-30:
For details on distance sensor / point lidar, see the discussion in the original PR here https://github.com/Microsoft/AirSim/pull/791 and the original feature request here https://github.com/Microsoft/AirSim/issues/405

The params of the distance sensor - pose , min_distance, max_distance all need to be still exposed, but we have an API for the default config. 

For those who want to do this by modifying code, you can directly params can be directly changed for any sensor in `https://github.com/Microsoft/AirSim/tree/master/AirLib/include/sensors/SENSOR_NAME/SENSOR_NAMESimpleParams.hpp`
For example, 
 - Barometer https://github.com/Microsoft/AirSim/blob/master/AirLib/include/sensors/barometer/BarometerSimpleParams.hpp 
- IMU - https://github.com/Microsoft/AirSim/blob/master/AirLib/include/sensors/imu/ImuSimpleParams.hpp
- Distance Sensor: https://github.com/Microsoft/AirSim/blob/master/AirLib/include/sensors/distance/DistanceSimpleParams.hpp

closed 2019-04-30 by madratman

rajatthosar, 2019-08-05:
I tried to implement the getDistanceSensorData(lidar_name, vehicle_name) method but I received the following error. 
NameError: name 'DistanceSensorData' is not defined

I tried to find the DistanceSensorData class but couldn't find it. Has anyone faced a similar issue? And if so, how did you resolve it? 
Any help is much appreciated

Hoshea7, 2019-08-14:
@rajatthosar 
I also meet this issue, have you solved it ?

ACLeighner, 2019-10-07:
To anyone looking for a fix to this.  I was able to get the distance sensor working by adding a definition of DistanceSensorData to the 'types' file: AirSim\PythonClient\airsim\types.py.

Just paste this in the bottom of that file:
class DistanceSensorData(MsgpackMixin):
    time_stamp = np.uint64(0)
    distance = Quaternionr()
    min_distance = Quaternionr()
    max_distance = Quaternionr()
    relative_pose = Pose()

ibrahim0v, 2019-12-31:
> For details on distance sensor / point lidar, see the discussion in the original PR here #791 and the original feature request here #405
> 
> The params of the distance sensor - pose , min_distance, max_distance all need to be still exposed, but we have an API for the default config. I'll add a PR to expose this params in settings.json soon.
> 
> For those who want to do this by modifying code, you can directly params can be directly changed for any sensor in `https://github.com/Microsoft/AirSim/tree/master/AirLib/include/sensors/SENSOR_NAME/SENSOR_NAMESimpleParams.hpp`
> For example,
> 
>     * Barometer https://github.com/Microsoft/AirSim/blob/master/AirLib/include/sensors/barometer/BarometerSimpleParams.hpp
> 
>     * IMU - https://github.com/Microsoft/AirSim/blob/master/AirLib/include/sensors/imu/ImuSimpleParams.hpp
> 
>     * Distance Sensor: https://github.com/Microsoft/AirSim/blob/master/AirLib/include/sensors/distance/DistanceSimpleParams.hpp
> 
> 
> Closing this now, but happy to keep the discussion alive. Please let us know if things are still not clear

@madratman are the settings.json distance sensor parameters out yet?