microsoft/AirSim #1030: How to build AirSim in a new scene

abc19850508, 2018-04-27:
I am trying to build AirSim in a new scene. I download  a scene called ZenGarden from Epic Games. And I built AirSim in ZenGarden following the same steps as building AirSim in LandscapeMountainsl. But I cant find the  car or multimotor when the program is running after building. And the camera flips here and there.

I want to find some help docs ( similiar to the docs unreal_custenv.md in AirSim/docs   to help build airsim in LandscapeMountains ) to help me build airsim in a new scene where there are many green trees and houses. Is there any url link of this kind?

If someone have this kind of experience , can you tell me which scene( it is better to be free ) fit and any kind of advice would be appreciated.

Thanks for your help!

Domi47, 2018-04-27:
To spawn drone or car in your new scenario, you only need to change Selected GameMode to AirSimGameMode as it's said on unreal_custenv.md tutorial

abc19850508, 2018-04-28:
thanks for your answer. I set the GameMode Override to AirSimGameMode.  And I thought AirSim worked in the new scenario. But I can not find the drone.  the camera switch without control. At the beginning there was no player start in this scenario. I add a player start and set it to different values to try to find the drone, but it didn't work.
[image]
[image]

Domi47, 2018-04-28:
Oh I see that scenario works different, I’m also new with unreal so don’t know how to change that. Are you using that scenario for any reason? If not, I’ve been working with another called SunTemple that is also free and works fine. Also, according to #998 they will release the Linux binaries for city environment soon.

sytelus, 2018-04-29:
You are probably talking about buying above environment which is available at https://www.unrealengine.com/marketplace/modular-neighborhood-pack

closed 2018-05-08 by sytelus