microsoft/AirSim #1032: Car simulation in City environment not showing properly via projectors

haider04, 2018-04-27:
Hi,
I am trying to run the the car simulation in city environment through 3 projectors (each projector connected to  Nvidia GeForce GTX 980) graphics cards in PC, but the image i am getting is not projecting on the whole curved screen. I can see the car in the screen centre and the car info on the complete left side of the projector as shown in the picture. 

Why i am having black screen in sides? It is running ok in laptop. I want to have the image projected on the whole screen, not just in the middle. 

I updated the latest Nvidia driver as well (397.31) but i am having the same issue. I am using the surround display option to span the image to full screen. 

Do i have to make some Nvidia settings or somewhere in .json file? 

Has anyone also experienced similar issue?

Regards,
Haider
[image]

sytelus, 2018-04-27:
This is likely related to how you are setting up GPUs to use which screens, or display settings. I don't think there is anything in AirSim itself that produces this behaviour.

haider04, 2018-05-02:
Hi,

I have set the 3 displays in surround display mode, making the 3 projectors to project a single image. The image through projectors work fine when i play some other simulation but when i open city environment in AirSim, the main screen is displayed only through the middle projector as shown in above picture, and the control values (throttle, brake etc) are shown on the left side (not visible in above picture). 

I was wondering if there is some resolution setting in AirSim which can make it project on the full screen by 3 projectors? Any clues?

Regards,
Haider

sytelus, 2018-05-08:
Are you saying other environment or other Unreal project works fine? AirSim doesn't do any setup for multiple screens. In full screen mode, Unreal Engine decides the layout and resolution based on your GPU capacity and settings.

closed 2018-10-17 by sytelus