microsoft/AirSim #1033: Recording Issue : AutoExposureSettings have no effect

michael-fonder, 2018-04-27:
I'm trying to record RGB pictures from the cameras. On the landscape map, it works well with default settings, although the color recorded pictures do not exactly match the one of the fpv view. However, when I'm running airsim in other environments (i tried the maps of ABoyAndHisKite and EpicZenGarden), the colors of the recorded pictures are completely off (either heavily over or underexposed) compared to the fpv view. I tried to adjust the recording settings, but all the "AutoExposure" parameters do not have any effect when I modify them. I found the issues #561 and #607 (and all the related ones) but they didn't help.

Can someone give me clues to work on to solve this issue? I would at least like to make those settings work and ideally to find a way to record pictures which have the exact same colors (i.e. postprocessing?) than what is displayed for the fpv view.

I have been investigating this issue for 3 days now. I found the part of the code where in PIPCamera where the settings are assigned to the capture element and understood that they are somehow related to the Content/HUDAssets in the unreal engine. Being new to UE4, I have however issues to understand how both interact and at which point in the code exactly airsim makes its render requests. Maybe helping me on this point could help me to find the issue?

michael-fonder, 2018-05-02:
Ok, I found the reason why the settings entered in the configuration file are not taken into account in the game engine. This is simply because the autoexposure checkboxes of the SceneCaptureComponent provided with the airsim plugin are not ticked by default. Once this is done, the setting are properly taken into account during the simulation.

[image]

james-mchugh, 2019-02-14:
@michael-fonder Can you please describe how to find these settings? I was unable to find the blueprint for BP_PipCamera,

michael-fonder, 2019-02-14:
@james-mchugh  Sure! First of all, you need to know that this method only works in the Unreal Editor. It means that this fix is impossible to implement on precompiled releases of Airsim.
Once you are in the editor, go to the content browser tab. On the bottom right, there is an eye, click on it and tick the "Show plugin content" box. Now in the tree on the left, you should see a directory named "Airsim", select it. The BP_PipCamera asset is located in the "Blueprint" directory. Just double-click on it to open the editor.

james-mchugh, 2019-02-14:
@michael-fonder Thank you! I will check this out when I get home!

madratman, 2019-02-14:
alternatively, you can also play with targetgamma (try something b/w 1.5 to 3). You can use this:

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1080,
        "Height": 720,
        "FOV_Degrees": 120,
        "TargetGamma": 1.5
      }
    ]
}
}
```

madratman, 2019-04-12:
closing as this was fixed in #1834

closed 2019-04-12 by madratman