microsoft/AirSim #1034: Error packaging Blocks project on linux

TritonSailor, 2018-04-27:
Packaging Blocks project on linux fails with the following error:

[273/305] Compile Module.EngineMessages.cpp
error: exception handling was disabled in PCH file but is currently enabled
error: definition of macro 'PLATFORM_EXCEPTIONS_DISABLED' differs between the precompiled header ('1') and the command line ('0')
2 errors generated.

How can I fix this?  I'm using Ubuntu 16.04, Unreal 4.16 and Airsim 1.10.   Log attached with more detail. 

[UnrealBuildTool-2018.04.26-23.52.21.txt](https://github.com/Microsoft/AirSim/files/1955594/UnrealBuildTool-2018.04.26-23.52.21.txt)

sytelus, 2018-04-27:
Try running clean.sh. Also see https://answers.unrealengine.com/questions/749732/cant-build-ue-project-on-ubuntu.html

kshitiz38, 2018-05-31:
This is similar to #802..You can see the workaround there.

madratman, 2020-08-11:
closing as outdated wrt current master. 
One can refer to the "packaging " section CI pipeline script - https://github.com/microsoft/AirSim/blob/master/pipelines/ci.yml

```
$(UE4_ROOT)/Engine/Build/BatchFiles/RunUAT.sh BuildCookRun \
                -project="$(pwd)/Unreal/Environments/Blocks/Blocks.uproject" \
                -nop4 -nocompile -build -cook -compressed -pak -allmaps -stage \
                -archive -archivedirectory="$(pwd)/Unreal/Environments/Blocks/Packaged/Development" \
                -clientconfig=Development -clean -utf8output
```

closed 2020-08-11 by madratman