microsoft/AirSim #1037: error when I run ./setup.sh

teoac, 2018-04-29:
I'm trying to build AirSim on Linux.
But I run ./setup.sh after cloning github, then I receive the below error messages.

---
...
Reading package lists... Done
E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial/main/binary-arm64/Packages  404  Not Found [IP: 91.189.88.149 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial-updates/main/binary-arm64/Packages  404  Not Found [IP: 91.189.88.149 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial-backports/main/binary-arm64/Packages  404  Not Found [IP: 91.189.88.149 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial-security/main/binary-arm64/Packages  404  Not Found [IP: 91.189.88.149 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
---

I guess that some file links are broken. How can I fix this problem?

sytelus, 2018-04-30:
Looks like you have issue with your Ubuntu installation or internet connection.

teoac, 2018-04-30:
Yes. I installed jetpack, NVcaffe, DIGITS before. I guess that version of some packages is not matched with the packages required to build AirSim.
I format my hard drive and install Ubuntu again. Finally, AirSim is well built.
Thank you for your reply.

closed 2018-04-30 by teoac