microsoft/AirSim #1038: Reducing header file includes for Unreal Engine classes

SvetoslavAngelov, 2018-04-29:
Hi Guys, 

I would like to contribute to the project, but I'm still making my way through the code base. While reading the code I noticed that there are few opportunities to reduce header includes for Unreal classes. For example in CarPawn.h:
```
#pragma once

// Use these:
#include "CoreMinimal.h"
#include "WheeledVehicle.h"
#include "SimJoyStick/SimJoyStick.h"
#include "CarPawnApi.h"
#include "CarPawn.generated.h"

// Instead of:
#include "CoreMinimal.h"
#include "WheeledVehicle.h"
#include "physics/Kinematics.hpp"
#include "CarPawnApi.h"
#include "SimJoyStick/SimJoyStick.h"
#include "UObject/ConstructorHelpers.h"
#include "Components/SkeletalMeshComponent.h"
#include "PhysicalMaterials/PhysicalMaterial.h"
#include "common/AirSimSettings.hpp"
#include "AirBlueprintLib.h"
#include "CarPawn.generated.h"

class UPhysicalMaterial;
class UCameraComponent;
class USpringArmComponent;
class UTextRenderComponent;
class UInputComponent;
class UAudioComponent;
```

The same can be done for the .cpp files, since there are a few includes, which aren't necessary. Would you be interested in a contribution of this type?

sytelus, 2018-04-30:
Yes, this would be a great contribution!

SvetoslavAngelov, 2018-04-30:
That's excellent! I'll go through a few more files and submit a pull request.

sytelus, 2018-10-17:
Thanks for this contribution!

closed 2018-10-17 by sytelus