microsoft/AirSim #1043: Freeze Option ?

Pierrolo, 2018-05-02:
Hi,

I am wondering if there is any way to freeze the simulation. I am asking because if I use an online RL algorithm then the quality of the simulation is very bad. What I mean by that is this:

```
i) I reset
ii) I get observations
iii) I get action based on those observations
iv) I apply this action for 1 sec
v) I get reward and new obersvations
(I update model based on reward and obs)
vi) I get the new action based on those  new observations
vii) I take new action for 1 sec
viii) I get reward and new obersvations
```
and so on...
My problem right now is that in the simulation, 1sec should pass by during `iv)` and `vii)`, so that the action I choose afterward it based on the observation right after that action, but now I choose the action based on the obs not right after the action but `1sec + computation_time` after the start of the action. And when I use `clockspeed = 5.0` that `computation_time` becomes very important

I hope this is clear enough.
What I want to know is: is there a way to freeze the simulation between 2actions, just like the simulation "freezes" in OPENAI  Gym, making it a really continuous process. 

Please let me know if my question is not clear enough,
Thanks :)

zbenic, 2018-05-06:
I am also interested in an answer to this question as I have the exactly the same problem.

sytelus, 2018-05-08:
Please take a look at Pause and Continue APIs: https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#pause-and-continue-apis

Pierrolo, 2018-05-09:
Hi,
Thanks, this is exactly what I am looking for. Unfortunately, it is not working. The example script [pause_continue_drone](https://github.com/Microsoft/AirSim/blob/master/PythonClient/pause_continue_drone.py) is not working. Return this kinf or error : 

[image]

Thanks for you help

zbenic, 2018-05-12:
Have you migrated to VS2017 nad Unreal 4.18 and rebuild everything?
I did that and I can get it to run.

My question is, when we pause the engine and later we resume, is the complete quadcopter state preserved, i.e. does the quadcopter resume the simulation with the exact linear and angular speeds and accelerations as it had in the moment of hitting pause?

sytelus, 2018-05-14:
Yes, I think @Pierrolo doesn't have the latest code. 

@zbenic - yes, complete state is preserved!

pienrojp, 2018-05-17:
I also get the same error as @Pierrolo  after do git pull a moment ago.
Do I need to rebuild AirSim again?

Thank you.

Pierrolo, 2018-05-17:
Hi, so I rebuild AirSim and updated to UE4. 18 and now it works just fine, maybe try doing that and it should be OK I figure

Pierrolo, 2018-05-28:
Hi,

So the pause options works for the drone but not for the environment, it doesn't pause the moving object... Anybody got an idea?

Very much appreciated,
Pierre

sytelus, 2018-10-17:
Pause and continue will only work for physics based object.