microsoft/AirSim #1044: .uproject files for released environments

NextSim, 2018-05-02:
If I am making modifications to AirSim, and want to include my generated plugin into one of the released projects, is there a way to do this?

From what I have read it would go into the plugin folder of the unreal project, but it looks like after downloading the released environments the unreal project isn't there, so I'm not sure if I am able to use the environments with my modifications

sytelus, 2018-05-08:
Our released environments uses copyrighted assets created by other artists so we can't distribute their source code. This would also, unfortunately, would prevent you to make any updates. However, many environments we use are available on Unreal Marketplace (for example, Modular Neighbourhood or Landscape Mountains or Zhang Jia Jie) so you can buy from there and make updates.

I've also asked this question to UE forums: https://answers.unrealengine.com/questions/788039/can-plugins-be-changed-for-packaged-project.html

NextSim, 2018-05-09:
Great, thanks

closed 2018-05-09 by NextSim