microsoft/AirSim #1046: API/Way to send signal to Unreal?

luis-bernard, 2018-05-02:
Hey there!

**Is there an API or a way to send a signal to Unreal from a python script?** (Something that allows me to change a variable in Unreal or that can be used as an event trigger).

I'm using AirSim for reinforcement learning (with the drone) and I need to reset the world for the different iterations of the learning process (by reseting the world I mean having the drone again in a start position ready to fly, as if it was launching for the first time).
I've already managed to do this using Unreal Engine blueprints but right now I'm pressing one of my keyboard keys as the event that triggers the reset. 

(Sorry in advance if this is not the right place to ask this question 🙂 )

NextSim, 2018-05-02:
https://github.com/Microsoft/AirSim/blob/master/docs/python.md

There is a reset api call you can use

https://github.com/Microsoft/AirSim/blob/master/docs/apis.md

sytelus, 2018-05-08:
It might be good to add API that calls specified Blueprint function somewhere.

RVBldr, 2018-11-21:
Internal VSTF 7432

jonyMarino, 2020-09-02:
related to #675