microsoft/AirSim #1048: Lock camera 3rd person view

niknil702, 2018-05-03:
Is it possible to lock the 3rd person view of the quadcopter (i.e. have a static view from behind which follows when the quadcopter rotates)?

I have tried to edit parameters in settings.json but have not managed to achieve what I'm after.

sytelus, 2018-05-03:
May be SpringArmChase works for you.  

https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#viewmode

Domi47, 2018-05-07:
I'm also interested in this feature. I see that using SpringArmChase camera is always on the same position (it follows the drone but keep always it initial direction), so it doesn't rotate with drone, what seems to be a bit uncomfortable. Is there any way to set the camera always behind the drone (so rotating with it)??

sytelus, 2018-05-08:
I would suggest to look in here and try setting different properties on spring arm:
https://github.com/Microsoft/AirSim/blob/c122d2efc746857c4cc1c503584a0981a6b14cf7/Unreal/Environments/Blocks/Plugins/AirSim/Source/CameraDirector.cpp#L10

Here's good doc on how to use spring arm: https://docs.unrealengine.com/en-us/Gameplay/HowTo/UsingCameras/SpringArmComponents

madratman, 2019-02-19:
Too late, but unreal cine camera is also worth checking out. https://docs.unrealengine.com/en-us/Engine/Sequencer/HowTo/CineCameraActors

closed 2020-09-02 by jonyMarino